Xilinx/QNN-MO-PYNQ #2: Question on "stride" of sliding window module

awai54st, 2018-03-15:
Hi,
Very impressive work! Thank you for sharing this project to us.
I have some questions regarding the behaviour of "stride" parameter in the sliding window.
Is it true that assigning 0 to "stride" parameter will result in a striding of 1 during convolution?
If so, does this module support a striding of 2? And what value should we assign to this stride parameter?
Best,
awai54st

giuliogamba, 2018-03-15:
Hi, 

In order to save DSP resources, in the sliding window the value of _stride_ passed to the function is actually log2(Standard_Stride). 
As you pointed out, when value is 0, the actual stride is 1. In order to have stride of 2 you should set the value to 1. 
BR

awai54st, 2018-03-16:
I see. Thank you!

awai54st, 2018-03-20:
Hi,
I was trying the CircularStreamingConvolutionInputGenerator function with a stride of 1 (i.e. stride of 2).
The input frame is a 1x7x7 square matrix with its data counting from 0 to 48. Kernel has size 3.
With a stride of 2, I think the correct output stream should be [0,1,2,7,8,9,14,15,16,2,3,4,9,10,11,16,17,18,...].
However, the result that I get is [0,1,2,7,8,9,14,1,2,16,17,18,23,24,25,30,31,32,...].

Here is part of my C simulation testbench:

```
#define MaxConvKernelDim_L1 3
#define MaxIFMChannels_L1 1
#define MaxIFMDim_L1 7
#define InpWidth_L1 8
#define MaxStride_L1 2
#define MaxOFMDim_L1 3

const unsigned int batch_size = 1;
const unsigned int Ker_DIM = 3;
const unsigned int In_DIM = 7;
const unsigned int In_CH = 1;
const unsigned int PadDim = 0;
const unsigned int Out_CH = 1;
const unsigned int Stride = 1;
const unsigned int PaddedinDim = In_DIM+2*PadDim;
const unsigned int Out_DIM = 3;//(PaddedinDim-Ker_DIM)/(1<<Stride)+1;

for (int i = 0; i < In_CH; i ++){
	for(int j =0; j < In_DIM*In_DIM; j++){
		AXI_VAL valIn;
		valIn = j;
		in_stream << valIn;
	}
}
CircularStreamingConvolutionInputGenerator<MaxConvKernelDim_L1, MaxIFMChannels_L1, MaxIFMDim_L1, MaxOFMDim_L1, InpWidth_L1, MaxStride_L1>(in_stream, out_stream, Ker_DIM, PaddedinDim, Out_DIM, Stride);

int counter_B = 0;
for (int j = 0; j < Ker_DIM*Ker_DIM*Out_CH * Out_DIM*Out_DIM; j ++){
	AXI_VAL valOut;
	out_stream.read(valOut);

	printf("result is %d \n", (int)valOut);
	counter_B ++;
}
```

Is there anything wrong with my testbench setting?

Best,
awai54st

giuliogamba, 2018-03-20:
Hi, 

unfortunately with that version, the value of Kernel%Stride has to be =0. 
So kernel=3 and stride=2 is not a possible combination. I'm going to push a version of the input generator with generic kernel and stride in the near future.

awai54st, 2018-03-20:
That would be extremely helpful, especially for those trying to implement mobilenet using your framework.

Thank you!

Best,
awai54st