Xilinx/QNN-MO-PYNQ #3: How to train on custom dataset?

nisssal, 2018-03-15:
Hi,

Could you please give us directions to train this to our own data set? 
Or is it possible to train this network on original yolo implementation and covert the .weights files to the required format? Any script?

giuliogamba, 2018-03-20:
Hi, 

I guess you are talking about Darknet training. First you will need to instrument Darknet to perform training on reduced precision weights and activations. The modified version used for this repo has not been open-sourced yet. Once the training on your dataset achieves the desired accuracy, you can export the .weights file into the .bin used in this example. The script to performe that will be released in the near future.

nisssal, 2018-03-21:
Thank you. The .weights conversion script will be really helpful.

riple, 2018-04-13:
What is the DL framework used in the training process for DoReFaNet? TensorFlow, Theano or MXNet?

giuliogamba, 2018-04-13:
[Tensorpack](https://github.com/ppwwyyxx/tensorpack) on top of TensorFlow has been used for DorefaNet

riple, 2018-04-16:
If you use different frameworks for the 2 demo networks, you must have different scripts to covert the weight parameters to a unified form for the FPGA. Is this right?
Will you share the training scripts and converting scripts in the future?

giuliogamba, 2018-04-16:
Yes, we have planned to release those scripts

blgpb, 2018-10-15:
Excuse me, when will it be released?

Ashsur, 2019-01-14:
Excuse me, when will it be released?

Kr0n0, 2019-03-06:
Hi. Any plans for the scripts release dates? Thx.

rafferino, 2019-08-05:
Hello, is the script still in progress or has the plan to release it to open-source changed?

ussamazahid96, 2019-08-20:
For quantized **dorefanet** training in **tensorpack** please have a look at [this](https://github.com/mohdumar644/QNN-MO-PYNQ) repo.

For quantized **dorefanet** training in **pytorch** please have a look at [this](https://github.com/mohdumar644/DoReFaNet-PyTorch) repo.

For quantized **TinyYolo** training (**pytorch**) please have a look at [this](https://github.com/mohdumar644/TinyYOLO-BNN)

Thanks!