Xilinx/QNN-MO-PYNQ #4: undefined symbol: forward_region_layer_pointer_nolayer

sallylee2017, 2018-03-28:
When I run tiny-yolo-image notebook on my pynq board I got this error. 

.... /darknet/libdarknet.so: undefined symbol: forward_region_layer_pointer_nolayer

There is no function names forward_region_layer_pointer_nolayer, how to fix that?

shinya321, 2018-03-28:
I'm troubled too, here is error message.

>   File "/opt/python3.6/lib/python3.6/ctypes/__init__.py", line 357, in __getattr__
>     func = self.__getitem__(name)
>   File "/opt/python3.6/lib/python3.6/ctypes/__init__.py", line 362, in __getitem__
>     func = self._FuncPtr((name_or_ordinal, self))
> AttributeError: /usr/lib/libdarknet.so: undefined symbol: forward_region_layer_pointer_nolayer
> Segmentation fault

it would be greatly appreciated if you could explain the details.
Best regards

sallylee2017, 2018-03-28:
I solved this problem. 
Initially I encounter this problem because I just git clone the QNN repo and installed the darknet myself instead of running setup.py in QNN folder. I found that the darknet from darknet website is different from the one in QNN. After I git clone the darknet from the address provided in setup.py and installed it, the libdarknet.so has the function forward_region_layer_pointer_nolayer! 

You can run nm -D to check if your libdarknet has the function forward_region_layer_pointer_nolayer. The one from darknet website only has forward_region_layer. 

Hope this can be helpful.

giuliogamba, 2018-03-28:
Yes, the installation should be performed using pip install, as shown in the documentation. 
This will automatically pull all the dependencies (like the modified darknet).

shinya321, 2018-03-28:
Thank you very much for your great advice!!
As you two mentioned, I just git clone the darknet from original website.
So after uninstalling darknet and installing the modified darknet( https://github.com/giuliogamba/darknet ),
everything went well!
```
layer     filters    size              input                output
    0 conv     16  3 x 3 / 2   416 x 416 x   3   ->   208 x 208 x  16
    1 conv     64  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  64
    2 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64
    3 conv     64  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x  64
    4 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    5 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128
    6 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    7 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256
    8 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
    9 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512
   10 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   11 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   13 conv    125  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 125
   14 detection
mask_scale: Using default '1.000000'
First layer SW implementation took 537473 microseconds
HW implementation took 137885 microseconds
Last layer SW implementation took 51984 microseconds
class: car      probability: 86%
class: dog      probability: 86%
class: bicycle  probability: 78%
```

I really appreciate your great job!

closed 2018-07-10 by giuliogamba

vscv, 2019-01-10:
Hi @shinya321 @giuliogamba 

Ran with no luck from PYNQ-Z2.

1. Using QNN-MO-PYNQ default installation with py.
# run /opt/darknet/python/darknet.py
1.1
net = load_net(b"cfg/tiny-yolo.cfg", b'./tiny-yolov1.weights',0)#b"tiny-yolo.weights", 0) 
only output [ ] 

1.2 
    #net = load_net(b"cfg/yolov2-tiny-voc.cfg", b'weight/yolov2-tiny-voc.weights',0)
# Segmentation fault
1.3
    #net = load_net(b"cfg/yolov2.cfg", b'./yolov2.weights',0)
# Segmentation fault

2. Using QNN-MO-PYNQ default installation with ./darknet
root@pynq:/opt/darknet# ./darknet detect yolov2-tiny.cfg yolov2-tiny.weights  data/dog.jpg
Segmentation fault
even change to yolov2-tiny-voc or yolov3-tiny also Segmentation fault.