Xilinx/QNN-MO-PYNQ #5: Kernel dies when trying to run tiny-yolo-image-loop

shwetad7, 2018-03-29:
Hi,

I have a pynq board with latest (2.1) SD card image into it. I have downloaded this (https://github.com/Xilinx/QNN-MO-PYNQ) repository and trying to run "tiny-yolo-image-loop" jupyter nottebook. Sometimes it runs perfectly but many times kernel dies in-between.
Is there a way to debug or analyse the issue.

giuliogamba, 2018-03-29:
Hi, 
Actually we never experienced that, but I assume it can be due to the device going out-of-memory. 
In general, you can check why a notebook kernel dies ny checking the log under `/var/log/jupyter.log`

MinLiAmoy, 2018-03-29:
same problem:D

shwetad7, 2018-03-30:
Hey giuliogamba,
Looks like some memory allocation/deallocation issue. Following is content copied from jupyter.log for your reference.

[I 00:21:18.864 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
[I 00:21:18.938 NotebookApp] 302 GET / (127.0.0.1) 4.79ms
[I 00:21:18.950 NotebookApp] 302 GET /tree? (127.0.0.1) 8.19ms
[W 00:21:34.353 NotebookApp] 404 GET /api/kernels/9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c/channels?session_id=3b9bb37d5bff4e0fbb0290deca8c577c (192.168.40.193): Kernel does not exist: 9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c
[W 00:21:34.445 NotebookApp] 404 GET /api/kernels/9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c/channels?session_id=3b9bb37d5bff4e0fbb0290deca8c577c (192.168.40.193) 213.01ms referer=None
[W 00:21:37.617 NotebookApp] Replacing stale connection: 9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c:3b9bb37d5bff4e0fbb0290deca8c577c
[W 00:22:06.471 NotebookApp] Replacing stale connection: 9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c:3b9bb37d5bff4e0fbb0290deca8c577c
[W 00:23:30.480 NotebookApp] Replacing stale connection: 9b30aaf4-0ba0-471d-a2f9-59ffe27fda6c:3b9bb37d5bff4e0fbb0290deca8c577c
[I 00:26:09.685 NotebookApp] Kernel started: 5e9785a8-c5b5-452f-aa6a-5c8ce6b80f21
[I 00:26:18.436 NotebookApp] Adapting to protocol v5.1 for kernel 5e9785a8-c5b5-452f-aa6a-5c8ce6b80f21
layer     filters    size              input                output
    0 conv     16  3 x 3 / 2   416 x 416 x   3   ->   208 x 208 x  16
    1 conv     64  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  64
    2 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64
    3 conv     64  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x  64
    4 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    5 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128
    6 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    7 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256
    8 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
    9 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512
   10 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   11 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   13 conv    125  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 125
   14 detection
mask_scale: Using default '1.000000'
[I 00:30:09.582 NotebookApp] Saving file at /qnn/HDMI_Yolo-withcam.ipynb
[I 00:32:09.594 NotebookApp] Saving file at /qnn/HDMI_Yolo-withcam.ipynb
layer     filters    size              input                output
    0 conv     16  3 x 3 / 2   416 x 416 x   3   ->   208 x 208 x  16
    1 conv     64  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  64
    2 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64
    3 conv     64  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x  64
    4 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    5 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128
    6 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    7 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256
    8 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
    9 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512
   10 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   11 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv    512  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x 512
   13 conv    125  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 125
   14 detection
mask_scale: Using default '1.000000'
*** Error in `/opt/python3.6/bin/python3.6': double free or corruption (out): 0x192fe008 ***
[I 00:34:54.661 NotebookApp] KernelRestarter: restarting kernel (1/5), keep random ports
kernel 5e9785a8-c5b5-452f-aa6a-5c8ce6b80f21 restarted

giuliogamba, 2018-03-30:
Hi, 

Is this happening in the original notebook? 
It seems you are calling the initialization of darknet `lib.parse_network_cfg` (performing memory allocation for the layers) twice. This might end-up in double memory allocation.

shwetad7, 2018-03-30:
But I have called lib.parse_network_cfg only once. Is it called internally in any other function ?
Is there a way to free this memory and allocate again?

giuliogamba, 2018-03-30:
You might want to check the call `free_network` in the darknet lib.

closed 2019-03-20 by giuliogamba