Xilinx/QNN-MO-PYNQ #6: test

DIAMONDWHILE, 2018-04-17:
This is a very powerful experiment.
Excuse me! in vivado hls , what should I do if I want to detect object for my model ?

giuliogamba, 2018-04-17:
Hi, 

I'm not sure what the question is. In general, if you want to train your own network, you will need to instrument Darknet to perform training on reduced precision weights and activations, or wait for the modified version used for this repo to be open-sourced.
Then youcan export the .weights file into the .bin used in this example. The script to perform that will also be released when the modified darknet will be released.

giuliogamba, 2018-04-25:
Hi, 

there is no explicit batch normalization in this implementation. We perform activation+batch normalization by mean of thresholds comparison as in [here](https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/qnn/src/library/hls/thresholding.h).

DIAMONDWHILE, 2018-04-27:
If I want to implement my MODEL on PYNQ
1. I will export the .weights file to .bin.
2. Execute the script.
Is that right ?