Xilinx/QNN-MO-PYNQ #7: Using webcam with QNN

arsam96, 2018-04-19:
Hey! Really thanks for the tutorial for QNN. I want to use my usb webcam to capture the image in real time and then use your model to identify it. Kidnly tell me where to change the code. I'm new to fpga domain.
Thanks.

Icarus0519, 2019-03-10:
Hi arsam96,

Recently, i want to achieve real-time object detection by QNN with pynq , and use usb webcam to capture the image, but my frame rate is only 1/4 FPS, that is need 4 seconds to load an image, it is too slow, no real-time, so i want to ask how about your code? what is your frame rate? I am new to fpga too, hope get some advise from you.

Hope to hearing from you soon, and thanks in advance!
Yuan Zhou.

giuliogamba, 2019-03-20:
Please refer to #34

closed 2019-03-20 by giuliogamba