Xilinx/QNN-MO-PYNQ #9: Tinier YOLO processing flow

joyliu37, 2018-06-26:
Hi,

I am wondering if this Tinier YOLO design have the same flow with Tincy YOLO shown in Xilinx website? 
https://forums.xilinx.com/t5/Xcell-Daily-Blog/Tincy-YOLO-a-real-time-low-latency-low-power-object-detection/ba-p/815840

giuliogamba, 2018-06-26:
Hi, 

yes, it's the same but the data reported in the Xcell blog are about the implementation on a different board, namely the [Ultra96](https://www.96boards.org/product/ultra96/).

joyliu37, 2018-06-26:
Thanks for your quick reply. Did you implement all the optimization? I found in the code you executing the first and last layer in python instead of the NEON acceleration. And maxpool is still in the architecture. 
Moreover, could tell me the compressed model's mAP and on PASCAL VOC.
Thank you so much!

giuliogamba, 2018-08-17:
It's the same model used on the other board, with 48.5 mAP on Pascal VOC. The optimization using NEON are not included in the current release, we do use a Python backend to evaluate first and last layer.

closed 2018-09-27 by giuliogamba