Xilinx/QNN-MO-PYNQ #10: HW rebuild for other plaform

joyliu37, 2018-07-02:
Hi, I am trying to rebuild this system for other platform, Could I change the device in make-hw.sh script to other device and redo the synthesis?

giuliogamba, 2018-07-09:
Hi, 

yes you can retarget the device in hls-syn.tcl, in the make-hw.sh script and the make-vivado-proj.tcl. 
Nonetheless, you're going to need to adapt also the the run-time SW running on the ARM cores, since the implementation is based on the XLNK driver for contiguous memory allocation.

joyliu37, 2018-07-10:
Thanks! I am trying to do the co-sim in HLS and run into the following error. Do you have any idea what's wrong with that? 

/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.cpp:55:1: error: uninitialized reference member ‘Layers::_network’ [-fpermissive]
In file included from /home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/vector:70:0,
                 from /home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.h:37,
                 from /home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.cpp:32:
/home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_inse
rt_aux(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const Layers::Layer&}, _Tp = Layers::Layer, _Alloc = std::allocator<Layers::Layer>, std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__no
rmal_iterator<Layers::Layer*, std::vector<Layers::Layer> >, typename std::_Vector_base<_Tp, _Alloc>::_Tp_alloc_type::pointer = Layers::Layer*]’:
/home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/bits/stl_vector.h:834:4:   instantiated from ‘void std::vector<_Tp, _Alloc>
::push_back(const value_type&) [with _Tp = Layers::Layer, _Alloc = std::allocator<Layers::Layer>, std::vector<_Tp, _Alloc>::value_type = Layers::Layer]’
/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.cpp:312:46:   instantiated from here
/home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/bits/vector.tcc:319:4: error: use of deleted function ‘Layers::Layer& Layer
s::Layer::operator=(const Layers::Layer&)’
In file included from /home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.cpp:32:0:
/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.h:63:16: error: ‘Layers::Layer& Layers::Layer::operator=(const Layers::Layer&)’ is implicitly deleted because the default definition would be ill-f
ormed:
/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.h:63:16: error: non-static reference member ‘Layers& Layers::Layer::parent’, can’t use default assignment operator
/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.h:63:16: error: non-static reference member ‘Network& Layers::Layer::network’, can’t use default assignment operator
In file included from /home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/bits/char_traits.h:41:0,
                 from /home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/ios:41,
                 from /home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/istream:40,
                 from /home/common/tools/Xilinx/Vivado/2017.4/lnx64/tools/gcc/bin/../lib/gcc/x86_64-unknown-linux-gnu/4.6.3/../../../../include/c++/4.6.3/fstream:40,
                 from /home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.h:36,
                 from /home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/library/host/layers.cpp:32:

make: *** [obj/layers.cpp_pre.cpp.tb.o] Error 1
ERROR: [COSIM 212-317] C++ compile error.
ERROR: [COSIM 212-321] EXE file generate failed.
ERROR: [COSIM 212-321] EXE file generate failed.
ERROR: [COSIM 212-331] Aborting co-simulation: C simulation failed, compilation errors.
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

giuliogamba, 2018-07-10:
Hi, 

for performance reasons, the host code embeds features of C++11, which is not completely supported in gcc 4.6.3. 
You can try to instrument the host code to avoid C++11 non supported features, like the one listed [here](https://gcc.gnu.org/gcc-4.6/cxx0x_status.html). 
Hope this helps.

joyliu37, 2018-07-10:
Um, make sense. Thanks!

joyliu37, 2018-07-11:
I fixed the previous compile problem and run the co-simulation. It seems another problem related to the axi interface pop up. Do you have any idea? Thanks!

/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3/sol1/sim/verilog/AESL_axi_master_hostmem2.v: Read request address                   16 exceed AXI master hostmem2 array depth:          1
$finish called at time : 1735 ns : File "/home/joey.liu/Work/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3/sol1/sim/verilog/AESL_axi_master_hostmem2.v" Line 695
## quit
INFO: [Common 17-206] Exiting xsim at Wed Jul 11 11:10:55 2018...
ERROR: [COSIM 212-303] Aborting co-simulation: RTL simulation failed.
ERROR: [COSIM 212-344] Rtl simulation failed.

giuliogamba, 2018-07-12:
Increase the depth of the m_axi to the size of the data moved into the accelerator. You have to increase the depth value in lines 149, 151 and 153 in the [top.cpp](https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/qnn/src/network/W1A3/top.cpp) file.

closed 2018-07-12 by giuliogamba