Xilinx/QNN-MO-PYNQ #15: how to rebuild this for new platform like zcu104

feolcn, 2018-09-16:
Hi, now i am working on zcu104, could you tell me that how to rebuild this for new platform ?
thank you very much ,now i can ran the PYNQ on my zcu104 board.next ,what should i do ?

feolcn, 2018-09-21:
@joyliu37 thank you so much, i will try it follow you said .

Pe3ucTop, 2018-10-03:
@joyliu37  
 How do you get through "lib_hw" compilation ? 
 I have error reporting "/usr/bin/aarch64-linux-gnu-ld: cannot find -lsds_lib"

giuliogamba, 2018-10-03:
If you're using the PYNQ2.3 image on ZCU104, please change `-lsds_lib` to `-lcma`

Pe3ucTop, 2018-10-03:
@giuliogamba Thanks, it helps.

Pe3ucTop, 2018-10-08:
@joyliu37 Back to question : Did you get HW acceleration working? Could you share result / files / compilation reports ? 
I still have no success.
Always getting "Bus error" on ZCU102 board at inference execution , and error in system logs : 
...
python3[4644]: unhandled alignment fault (7) at 0x7f90757080, esr 0x92000061, in libc-2.27.so[7fa3f98000+140000]
[  150.387932] CPU: 2 PID: 4644 Comm: python3 Not tainted 4.14.0-xilinx-v2018.2 #1
[  150.387934] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
...

And do not know where to look for problem source.
At the same time I have BNN working, but it differ in HW structure (use single AXI data connection).

Pe3ucTop, 2018-10-09:
Probably it's because of migration from 32 bit arch to 64 bit.
It's look like problem is in data/functions alignment...

giuliogamba, 2018-10-22:
Just pushed the update to 2.3 and support to Ultra96 PYNQ board (with Ultrascale+ Zynq)

closed 2018-10-30 by giuliogamba

jason911ss, 2019-04-22:
@giuliogamba @feolcn @joyliu37 @Pe3ucTop Could you give some help about #39  ?
Thank you very much!!

jamesthesken, 2019-06-05:
@giuliogamba Got the same error, `Only supported on a Ultra96, Pynq-Z1 or Pynq-Z2 Board` How might I go about rebuilding? 

Thank you for any help.

giuliogamba, 2019-06-06:
@jamesthesken Yes, those are the boards we currently support. If you want to port to 104, you should change the target part in both vivado_hls and vivado to `xczu7ev-ffvc1156-2-e`.

jamesthesken, 2019-06-06:
@giuliogamba Thank you for the response! I was worried I would run into a dead end.

This is my first round with FPGA development so please bare with me. Could this be accomplished by following the steps outlined in the Hardware design rebuilt section of the readme? 

I have a license for Vivado Design Suite, I would need to install this as well as the Vivado High Level Synthesis?

Thank you for your help!

jamesthesken, 2019-06-06:
After doing some reading, what I need to do by following the instructions outlined in README.md, is on 

**Step 4**:  `./make-hw.sh {network} {platform} {mode}`

For example if I want to test tinier yolo on my ZCU104, I would change this command to:
`./make-hw.sh W1A3 xczu7ev-ffvc1156-2-e a`

Or do I instead modify [make-hw.sh](https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/qnn/src/network/make-hw.sh) to include the target `xczu7ev-ffvc1156-2-e`

Thank you!

jamesthesken, 2019-06-11:
Hi @giuliogamba is there an estimate for official support for the ZCU104?

Thank you.

jamesthesken, 2019-06-20:
Just wanting to follow up, I have been trying the rebuild script but with no luck.

jamesthesken, 2019-07-18:
@giuliogamba