Xilinx/QNN-MO-PYNQ #18: Python3 to lib_hw.so connection problem on ZynqMP (aarch64) , "python3: unhandled alignment fault (7)"

Pe3ucTop, 2018-10-09:
I compile this library for ZCU102 board (ZynqMP Ultrascale+) and get working app_hw.elf binary.
When trying to use library from Python3.6 getting error (python crash) with:
In console, when python crash/get back to shell:
"Bus error"
In system logs:
" 
python3: unhandled alignment fault (7) at 0x7f78695080, esr 0x92000061, in libc-2.27.so[7f8bed6000+140000]
CPU: 0 PID: 5065 Comm: python3 Not tainted 4.14.0-xilinx-v2018.2 #1
Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
"

feolcn, 2018-10-11:
> Official support to Ultrascale+ PYNQ boards (Ultra96/ZCU104) will be added soon.

Good news,and when it release?

Pe3ucTop, 2018-10-16:
Is there any news ?

giuliogamba, 2018-10-22:
Just pushed the update to 2.3 and support to Ultra96 PYNQ board.

giuliogamba, 2018-10-22:
Just pushed the update to PYNQ 2.3 and support to Ultra96 PYNQ board (with Ultrascale+ Zynq)

Pe3ucTop, 2018-11-05:
@giuliogamba 
Sorry for long reply, was out of the office.
Update did not solve problem, still getting "Bus error" with "unhandled alignment fault (7)"

giuliogamba, 2018-11-06:
It seems odd, we were able to compile everything on Ultra96 board. Are you sure you pulled the latest version? On which board are you experiencing this?

Pe3ucTop, 2018-11-06:
Yes, I tried the latest version too, without success.
Yes, It's odd, because app_hw work without problems, but lib_hw fail , both use same functions.
But I find problem source, it is "memset" which is for (char *)type buffer , in file "qnn/src/library/host/offload-utils.cpp" , it is used in "padTo" function.
It failing when zeroing testBuffer provided in singleInference() under lib_hw . If it is executed on additional buffer, it do not fail.
But final solution I use is in file "qnn/src/library/host/offload-utils.cpp" line 40 :
"OffloadUtils::memset(bufferPadded, 0, outputSize); " 
need replace with
"OffloadUtils::memset((ExtMemWord *)bufferPadded, 0, outputSize);"

I testing on 2 different boards: Xilinx ZCU102 and Enclustra Mars XU3 .
And using lib_hw from Python3.6 for quicker prototyping.

giuliogamba, 2018-11-07:
You are correct, the problem is in the memset when used on non-cacheable memory with `-O3` as optimization strategy. 
That's why we changed the optimization strategy [here](https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/qnn/src/library/host/Makefile) to `-O2` and avoid the problem.

Pe3ucTop, 2018-11-09:
Yes, you are right, optimization strategy changes result working lib_hw.
Thanks.

closed 2018-11-10 by Pe3ucTop