Xilinx/QNN-MO-PYNQ #21: hw rebuild

lyn2333, 2018-11-09:
My board is PYNQ-Z2. When I implement the project, something wrong happens. How could I deal with it?

The follow is errors:
[Place 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 62315 of such cell types but only 53200 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.

[image]

giuliogamba, 2018-11-09:
Duplicate #11, supported Vivado version is 2017.4

lyn2333, 2018-11-09:
I have one question: if I rebuilt the project  using vivado 2017.4 and do some modifications, then generate bitsreams, could I use the img v2.3 to call its functions? I find something in "https://pynq.readthedocs.io/en/latest/pynq_sd_card.html#building-the-image" like this:
[image]

giuliogamba, 2018-11-09:
Yes, you can

closed 2018-11-09 by giuliogamba

lyn2333, 2018-11-10:
I am so sorry to disturb you again.
    The utilization ratio of BRAM is too high, which accounts for about 99% . I have no room for revision and I didn't find the IP --- Block Memory Generator. Could you tell me where to decrease it? 
Thank you.

lyn2333, 2018-11-10:
[image]
[image]