Xilinx/QNN-MO-PYNQ #24: pip3 install problem

chiayilin1991, 2018-12-17:
My board is pynq-z2.
And, I tried to pip3 install git+https://github.com/Xilinx/QNN-MO-PYNQ.git, 
but there was no any response after cloning.

Collecting git+https://github.com/Xilinx/QNN-MO-PYNQ.git
  Cloning https://github.com/Xilinx/QNN-MO-PYNQ.git to /tmp/pip-bl5011_8-build

I had to wait about 20min.
By the way, I could ping github.com.

giuliogamba, 2019-01-02:
You can try to first clone the repository on the board, then move to the folder and use the command `pip3 install ./`. The installation process is quite slow, and in the non-verbose mode not many information is given to the user about the status. You can try to execute pip3 with the flag `-v` to increase verbosity and check the installation status.

giuliogamba, 2019-01-17:
@chiayilin1991 Where you able to solve the problem?

closed 2019-02-19 by giuliogamba