Xilinx/QNN-MO-PYNQ #26: How to get the boxes, classes and probs, it seems only plot out

vscv, 2019-01-09:
Hi, giuliogamba
After ran the tiny-yolo-image.iynb, only get the plotted image with bbox and a probs file. 
How to get each boxes information like (meta.names[i], probs[j][i], (boxes[j].x, boxes[j].y, boxes[j].w, boxes[j].h) as /opt/darknet/python/darknet.py.

Thanks.

giuliogamba, 2019-01-17:
As you see in the notebook [example](https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/notebooks/tiny-yolo-image.ipynb), the notebook is calling the `draw_detection_python` function from darknet, which is defined [here](https://github.com/giuliogamba/darknet/blob/master/examples/detector.c). 
If you want to get more information, you can modify the call to return the variable boxes, probs etc in darknet, recompile the darknet shared object in order to get those info in python. 
Hope this helps.

closed 2019-03-20 by giuliogamba

Icarus0519, 2019-03-23:
> Thanks for your help.

Hi vscv,

how about you get the boxes information? I have the same problem as you, i need to get the detection boxes coords, but i dont know how to print it on my jupyter, can you give me some suggestions?

kind regards,
Icarus.

vscv, 2019-04-09:
> > Thanks for your help.
> 
> Hi vscv,
> 
> how about you get the boxes information? I have the same problem as you, i need to get the detection boxes coords, but i dont know how to print it on my jupyter, can you give me some suggestions?
> 
> kind regards,
> Icarus.

Hi lcarus, 

No luck recently. As giuliogamba says, you need to modify the c code then recompile yolo, but it never success in my PYNQ. I may continue try it when i have time.

:<

awais980, 2020-07-13:
> Hi lcarus,
>
> No luck recently. As giuliogamba says, you need to modify the c code then recompile yolo, but it never success in my PYNQ. I may continue try it when i have time.
>
> :<

Any success to get bounding boxes, etc. till now?

Icarus0519, 2020-07-15:
> Thanks for your help.
>
> Hi vscv,
> how about you get the boxes information? I have the same problem as you, i need to get the detection boxes coords, but i dont know how to print it on my jupyter, can [...]

Hi awais980,

Have you completed the modification of c code? but then didnt success?
After completing the modification of the c code, I can successfully get the detection boxes coords, only need to add a few code. Unfortunately, because it is too long, I really cannot remember exactly how to do it, I am so sorry, but I can sure this can be achieved.
Good luck. (*^_^*)