Xilinx/QNN-MO-PYNQ #28: No Output When Using QNN-MO-PYNQ default installation

vscv, 2019-01-10:
Hi @shinya321 @giuliogamba 

Ran with no luck from PYNQ-Z2.

1. Using QNN-MO-PYNQ default installation with py.
 run /opt/darknet/python/darknet.py
1.1
net = load_net(b"cfg/tiny-yolo.cfg", b'./tiny-yolov1.weights',0)#b"tiny-yolo.weights", 0) 
only output [ ] 

1.2 
net = load_net(b"cfg/yolov2-tiny-voc.cfg", b'weight/yolov2-tiny-voc.weights',0)
 Segmentation fault

1.3
net = load_net(b"cfg/yolov2.cfg", b'./yolov2.weights',0)
Segmentation fault

2. Using QNN-MO-PYNQ default installation with ./darknet
root@pynq:/opt/darknet# ./darknet detect yolov2-tiny.cfg yolov2-tiny.weights  data/dog.jpg
Segmentation fault
even change to yolov2-tiny-voc or yolov3-tiny also Segmentation fault.

_Originally posted by @vscv in https://github.com/_render_node/MDEyOklzc3VlQ29tbWVudDQ1MzAwNTY5Mg==/timeline/issue_comment#issuecomment-453005692_

giuliogamba, 2019-01-10:
Are you executing as root?

vscv, 2019-01-11:
> Are you executing as root?

Yes, is root, but let me check again.

giuliogamba, 2019-01-17:
Are you executing this via terminal? Are the example notebooks working in your setup?

closed 2019-03-20 by giuliogamba