Xilinx/QNN-MO-PYNQ #32: Error when i install the QNN package to my board

yupanpanwangrui, 2019-03-01:
When i use 
     sudo pip3 install git+https://github.com/Xilinx/QNN-MO-PYNQ.git
to install the QNN package to my board PYNQ Z2, the following error exists:

The directory '/home/xilinx/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executi.
The directory '/home/xilinx/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip .
Collecting git+https://github.com/Xilinx/QNN-MO-PYNQ.git
  Cloning https://github.com/Xilinx/QNN-MO-PYNQ.git to /tmp/pip-cz7ywm9o-build
Installing collected packages: qnn-loopback
  Running setup.py install for qnn-loopback ... error
    Complete output from command /opt/python3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-cz7ywm9o-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().repl:
    Running pre installation script...
    Building hardware library...
    Cloning into '/tmp/pip-cz7ywm9o-build/qnn/src/library/rapidjson'...
    Note: checking out 'tags/v1.1.0'.
    
    You are in 'detached HEAD' state. You can look around, make experimental
    changes and commit them, and you can discard any commits you make in this
    state without impacting any branches by performing another checkout.
    
    If you want to create a new branch to retain commits you create, you may
    do so (now or later) by using -b with the checkout command again. Example:
    
      git checkout -b <new-branch-name>
    
    HEAD is now at f54b0e4... Merge pull request #714 from miloyip/version1.1.0
    /usr/bin/ld: cannot find -lcma
    collect2: error: ld returned 1 exit status
    make[1]: *** [output/lib_hw.so] Error 1
    make: *** [lib_hw] Error 2
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-cz7ywm9o-build/setup.py", line 56, in <module>
        subprocess.check_output(["make", "-j2", "-C", my_path + "qnn/src/network/", "lib_hw"])
      File "/opt/python3.6/lib/python3.6/subprocess.py", line 336, in check_output
        **kwargs).stdout
      File "/opt/python3.6/lib/python3.6/subprocess.py", line 418, in run
        output=stdout, stderr=stderr)
    subprocess.CalledProcessError: Command '['make', '-j2', '-C', '/tmp/pip-cz7ywm9o-build/qnn/src/network/', 'lib_hw']' returned non-zero exit status 2.
    
    ----------------------------------------
Command "/opt/python3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-cz7ywm9o-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.clos/

Icarus0519, 2019-03-03:
what is your image version？

yupanpanwangrui, 2019-03-04:
v2.2

Icarus0519, 2019-03-04:
maybe you can try 2.3