Xilinx/QNN-MO-PYNQ #34: Implementation of real-time object detecion

Icarus0519, 2019-03-10:
Hi, giuliogamba,

Really thanks for your QNN tutorial ,recently, i want to achieve real-time object detection by QNN with pynq, i changed the tiny-yolo-image-loop.ipynq code, capture the image through usb webcam in the loop because i want to achieve real-time detection in this way, but my frame rate is only 1/4 FPS, that means it takes 4 seconds to load an image, it is too slow, no real-time, so i want to ask how can i improve my frame rate? I am new to pynq domain, hope get some advise from you.

Hope to hearing from you soon, and thanks in advance!
Icarus.

giuliogamba, 2019-03-20:
I would suggest you to read [this paper](https://export.arxiv.org/pdf/1806.08085). 
In there we explain how we were able to achieve 18 fps on the Ultra96 board. The hardware used is the same you can find in this project, but we haven't released the integration with darknet. 
Hope that work will provide insight on how to achieve what you want.

closed 2019-03-20 by giuliogamba

KaTaiHo, 2019-04-27:
What is the expected fps for an optimized pynqz1-z2?