Xilinx/QNN-MO-PYNQ #35: training model

Changyiyu, 2019-03-11:
Hello,
Can I get the training model of dorefanet-classifier?   I want to train the plant classes.

ussamazahid96, 2019-03-28:
you should take a look at this [repo](https://github.com/mohdumar644/QNN-MO-PYNQ)

KaTaiHo, 2019-04-27:
@ussamazahid96 were you able to successfully train new weights for QNN? If so, could you please share the steps to do so? I have looked at the repo that you mentioned above but couldn't get the npz files needed etc.

mohdumar644, 2019-04-27:
@KaTaiHo did you follow the steps at https://github.com/mohdumar644/QNN-MO-PYNQ/tree/master/qnn/src/training  README?

KaTaiHo, 2019-04-27:
@mohdumar644 Sorry I made a mistake in my previous post. I meant to ask if we needed to update the npy files such as tinier-yolo-conv0-W.npy from https://github.com/Xilinx/QNN-MO-PYNQ/blob/master/notebooks/tiny-yolo-image.ipynb.

mohdumar644, 2019-04-28:
Yes if training custom, they are needed for PS layers, only in QNN-MO-PYNQ (not BNN).

Asulr, 2020-03-25:
> @KaTaiHo did you follow the steps at https://github.com/mohdumar644/QNN-MO-PYNQ/tree/master/qnn/src/training README?
> 
> If you do not use tensorpack, also see https://github.com/mohdumar644/DoReFaNet-PyTorch

Hi, sorry to bother you. I have viewed the comments above and I have ran the DoreFaNet-PyTorch,and got the npz file. But when I use bin-gen from https://github.com/mohdumar644/QNN-MO-PYNQ/tree/master/qnn/src/training, I was failed. There don't have the file 0.npy.

B1D1ng, 2020-07-22:
> Hi, sorry to bother you. I have viewed the comments above and I have ran the DoreFaNet-PyTorch,and got the npz file. But when I use bin-gen from [...]

Hi! sorry to bother you. Have you solved this problem and succeed training your custom dataset?

B1D1ng, 2020-07-22:
> @KaTaiHo did you follow the steps at https://github.com/mohdumar644/QNN-MO-PYNQ/tree/master/qnn/src/training README?
> 
> If you do not use tensorpack, also see https://github.com/mohdumar644/DoReFaNet-PyTorch

Hi, sorry to bother you. If I want to train my own custom dataset for the QNN-MO-PYNQ, I should do the steps in this README?