Xilinx/QNN-MO-PYNQ #45: Hardware Rebuild Utilization exceeds 100%

dustinjoe, 2019-09-08:
Hello.
I am trying to rebuild hardware.
I notice that hardware utilization reports exceed 100% for some device parts.
For example, for the W1A2 ultra96:
== Utilization Estimates
================================================================
* Summary: 
+-----------------+---------+-------+--------+-------+-----+
|       Name      | BRAM_18K| DSP48E|   FF   |  LUT  | URAM|
+-----------------+---------+-------+--------+-------+-----+
|DSP              |        -|      -|       -|      -|    -|
|Expression       |        -|      -|       0|     47|    -|
|FIFO             |        -|      -|       -|      -|    -|
|Instance         |      150|     67|   26404|  72210|    -|
|Memory           |      256|      -|    4096|    384|    -|
|Multiplexer      |        -|      -|       -|   2445|    -|
|Register         |        -|      -|     511|      -|    -|
+-----------------+---------+-------+--------+-------+-----+
|Total            |      406|     67|   31011|  75086|    0|
+-----------------+---------+-------+--------+-------+-----+
|Available        |      432|    360|  141120|  70560|    0|
+-----------------+---------+-------+--------+-------+-----+
|Utilization (%)  |       93|     18|      21|    106|    0|
+-----------------+---------+-------+--------+-------+-----+

LUT 106%, does this matter? I am using 2017.4 on ubuntu.
Actually I also tried on 2018.3 on another machine and exceeds as well.
What might be the reason for these? Are there any additional settings I need to make?
Thank you!

toannv123, 2019-09-09:
I have the same problem

ERROR: [Place 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 59705 of such cell types but only 53200 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.

toannv123, 2019-09-10:
I found that the problem for W1A2 ultra96 has been fixed. 

However, it is still FAILED for pynqZ1-Z2.

Could you help me check it? Thank you so much.

%%%%%%%%%%%%%%%%
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device
ERROR: [Place 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 59705 of such cell types but only 53200 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [Place 30-640] Place Check : This design requires more LUT as Memory cells than are available in the target device. This design requires 31215 of such cell types but only 17400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [Place 30-640] Place Check : This design requires more LUT as Distributed RAM cells than are available in the target device. This design requires 29892 of such cell types but only 17400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [Place 30-640] Place Check : This design requires more RAMS64E cells than are available in the target device. This design requires 19968 of such cell types but only 17400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
INFO: [Timing 38-35] Done setting XDC timing constraints.
Phase 1.2 IO Placement/ Clock Placement/ Build Placer Device | Checksum: dfeb4d7c

giuliogamba, 2019-09-10:
Which version of Vivado are you using? The hardware build has been tested with 2017.4

toannv123, 2019-09-10:
I used Vivado version 2018.3.

I remember that the hardware was built successfully with vivado 2018.3 and the following source code:

commit b41cf6ec2b6d7c9e68bda0d988e2eb3223d3f2a6
Author: Giulio <giuliog@xilinx.com>
Date:   Mon Oct 22 13:57:11 2018 +0100

toannv123, 2019-09-10:
For more information, utilization of hardware resource is estimated by Vivado 2018.3 as follows (142%):

== Utilization Estimates
================================================================
* Summary: 
+-----------------+---------+-------+--------+-------+
|       Name      | BRAM_18K| DSP48E|   FF   |  LUT  |
+-----------------+---------+-------+--------+-------+
|DSP              |        -|      -|       -|      -|
|Expression       |        -|      -|       0|     32|
|FIFO             |        -|      -|       -|      -|
|Instance         |      214|     58|   33349|  73006|
|Memory           |      256|      -|    4096|    384|
|Multiplexer      |        -|      -|       -|   2427|
|Register         |        -|      -|     513|      -|
+-----------------+---------+-------+--------+-------+
|Total            |      470|     58|   37958|  75849|
+-----------------+---------+-------+--------+-------+
|Available        |      280|    220|  106400|  53200|
+-----------------+---------+-------+--------+-------+
|Utilization (%)  |      167|     26|      35|    142|
+-----------------+---------+-------+--------+-------+

dustinjoe, 2019-09-10:
thank you for reply
let me redownload this repo and try again

giuliogamba, 2019-09-11:
Please use 2017.4, which has been used to generate the bitstreams in this repo. 
Please note that you're also using the Vivado HLS estimates to draw conclusions, but those HLS reports are indeed estimates, so not 100% accurate.

closed 2019-09-11 by giuliogamba