Xilinx/QNN-MO-PYNQ #57: Errors rebuilding hardware design

B4k3, 2020-05-28:
I'm trying to rebuild the hardware design, but i get the following error 

> INFO: [HLS 200-10] Cleaning up the solution database.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "
INFO:` [Common 17-206] Exiting vivado_hls at Thu May 28 09:41:28 2020...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

I'm using Vivado 2017.4 on a Ubuntu installation.

Thanks in advance

giuliogamba, 2020-05-28:
The error message here attached is not very clear. Can you please open the generated HLS project from the GUI and launch csim again? The error message could be more revealing.

B4k3, 2020-05-29:
Hi, i have launched it from the GUI,  this is the error i have got:

> Starting C simulation ...
/opt/Xilinx/Vivado/2017.4/bin/vivado_hls /home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3-pynqZ1-Z2/sol1/csim.tcl
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2017.4/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'user' on host 'user-All-Series' (Linux_x86_64 version 5.3.0-51-generic) on Fri May 29 11:48:49 CEST 2020
INFO: [HLS 200-10] On os Ubuntu 18.04.4 LTS
INFO: [HLS 200-10] In directory '/home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn'
INFO: [HLS 200-10] Opening project '/home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3-pynqZ1-Z2'.
INFO: [HLS 200-10] Opening solution '/home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 2ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source /home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3-pynqZ1-Z2/sol1/csim.tcl"
    invoked from within
"hls::main /home/user/workspace/QNN-MO-PYNQ/qnn/src/network/output/hls-syn/W1A3-pynqZ1-Z2/sol1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C simulation.

zGitHubj, 2021-12-20:
**BEST SOLUTION**
Hello,
I have seen the same issue with ubuntu 18.4 and Vivado 19.1.
I found this answer:
https://forums.xilinx.com/t5/Design-Entry/Vivado-2018-3-error-launch-hls/td-p/940856
So, after installing build-essential the issue gone.

url: https://support.xilinx.com/s/question/0D52E00006hplQGSAY/hls-20183-csim-is-broken-on-ubuntu-1804-lts?language=en_US